airalab/sensors.social #36: Частые провалы в сохранении данных

ip-airalab, 2021-08-07:
При текущей архитектуре данные регулярно теряются и недоступны в истории.
6.08, около 20:00 пропали данные
[image]

07.08, около 11:35 публичная нода была перезагружена, однако данные за период не сохранились.
[image]

Данные постоянно отслеживаются экологами, поэтому провалы в истории очень неприятны.
[image]

Надо подумать над совершенствованием архитектуры. Что надо сделать? запустить дополнительную ноду? Отправлять данные напрямую в IPCI? Настроить автоматический перезапуск публичной ноды? Нужно решение для повышения надежности.

tubleronchik, 2021-08-12:
- [ ] Распараллерить потоки

- [ ] Добавить обработку ошибок —исключений (на моменте отправки икстренсика)

- [ ] Сделать описание обработки исключений

- [ ] Менеджер очередей

- [ ] Скрипт с настройкой среды по чистке логов раз в неделю

- [ ] Параллельные потоки логгирования

tubleronchik, 2021-08-17:
Основные проблемы по сохранению истории устранены. В случае неотправки данных по какой-либо причине, они должны будут появиться через сутки. Неотправленные данные за несколько прошлых дней будут добавлены в ближайшее время.

closed 2021-08-18 by Vourhey