airalab/sensors.social #55: выгрузка 31.10-05.11

olga23list, 2021-11-05:
прошу проверить выгрузку по всем датчикам, по многим датчикам нет 04/11
по датчику sensor_id:
 83f93506e8056e9b5a0992e11dfebeb6318de3a1a262acbc50c2759c4168f9ea
нет 03/11 и 04/11
07caa8e104dd944a73301deb0557ddcd2fc7db1956388c523bda0885b13c1073
выгрузка заканчивается 03/11 в 09-04

olga23list, 2021-11-05:
4. ID бd4a4dbc616245672294ef9f0794de2cfdd439e32fd6c4142a23109c7f5f8e8b8 – НЕТ ДАННЫХ

5. ID 2d8621681df30ee862dafbea275683f1037ec7983d0116725b549a20f01dac45 – НЕТ ДАННЫХ

7. ID 7c499116a58e2321efceb71f5f2d6005e06e6b83d8cc2dae40dda07fb283975e - НЕТ ДАННЫХ ЗА 31,10 И 01,11

9. ID 21148bf2bd0830c440e397b7d17af7feab0e5d0469d8f1759e886a1fe7061e80 – НЕТ ДАННЫХ

11. ID 652768ac54dac2b73f6cf9c5a3146e895018a067eab6eee226b2f39073d2af5a - НЕТ ДАННЫХ

12.  ID 78adf6508f0d0be98baae558f3e9049a0abcd81300f4f3997506f5658c40fff2 –  НЕТ 04/11 И 05/11

15. ID 5a891e7ecb5044e8309fb6180cfa8d2ddbc2036615f13166f9f69becab8c4ed2 – НЕТ ДАННЫХ

18. ID debd6d21b15c8caa69e39a1b82b7e5d515c887af322cb1a356216e5d2283b395 –  ВЫГРУЗКА С 12-11 (31/10)

23. ID 83f93506e8056e9b5a0992e11dfebeb6318de3a1a262acbc50c2759c4168f9ea –  02/11 ДО 16-00, НЕТ ДАННЫХ ЗА 03/11 И 04/11

25. ID 07caa8e104dd944a73301deb0557ddcd2fc7db1956388c523bda0885b13c1073 -  НЕТ ДАННЫХ ЗА 31/10, 01/11 ВЫГРУЗКА С 22:16, 03/11 ВЫГРУЗКА ДО 09:04, ЗА 04/11 И 05/11 ДАННЫХ НЕТ

26. ID 34ac74c096ebd91fc250f9acebb8f339872151dc35ce160f8a41428f38087e84 – НЕТ ДАННЫХ

tubleronchik, 2022-10-19:
Не актуально.

closed 2022-10-19 by tubleronchik