airalab/sensors.social #61: выгрузка за 05.01.2022

olga23list, 2022-01-07:
из 34 датчиков в выгрузке только 16 
[image]

[05.01.csv](https://github.com/airalab/sensors.robonomics.network/files/7830245/05.01.csv)

olga23list, 2022-01-12:
sensor_id:   d4a4dbc616245672294ef9f0794de2cfdd439e32fd6c4142a23109c7f5f8e8b8 - с 01/01 по 05/01 данных нет
sensor_id:   7c499116a58e2321efceb71f5f2d6005e06e6b83d8cc2dae40dda07fb283975e - с 01/01 по 05/01 данных нет
sensor_id: 15cb0549a4c80e75f865c41f9079067b848e4bf2c9ce2258f103449a8b754165 -с 01/01 по 05/01 данных нет
sensor_id:  83f93506e8056e9b5a0992e11dfebeb6318de3a1a262acbc50c2759c4168f9ea  -с 01/01 по 05/01 данных нет
sensor_id:  429efb5ac6b9328bca767a18c763d39457de30737b497c4aff439d46d4c68cb5 - данные только за 01/01
sensor_id: 34ac74c096ebd91fc250f9acebb8f339872151dc35ce160f8a41428f38087e84 - данные только за 05/01
sensor_id: 35c9797b45154048bb004d912906f53d16906b2cb67c0afe9fd33c257688ad57 данные за 6 и 7, много нулевых значений
sensor_id:   62436b6ded848ce5c7f4ae62b911b9f9cd61b94a1d62c141822ea756b5121f4a - с 01/01 по 05/01 данных нет
sensor_id:  84ee4d5a45534151de24edb04ac164e6763a8dbd3a99755002d63c2fed2f3e90 - с 01/01 по 05/01 данных нет

при выгрузке данных с 01/01 по 08/01 данные за 1, 2, 3, 4, 5 января отсутствуют

tubleronchik, 2022-01-12:
На двух серверах был крупный сбой, в связи с чем часть данных была утеряна.

closed 2022-01-12 by tubleronchik