airalab/sensors.social #88: Нет кнопки скачать историю

ip-airalab, 2022-10-11:
Предлагаю добавить на главную страницу функционал скачивания истории.
вариант 1. Ссылка скачивания за период https://sensors.robonomics.network/#/download размещается в верхней строке.
вариант 2. Ссылка скачивания за выбранную для отображения на странице дату на странице размещается возле календаря в нижней строке (тогда не получится скачать недельный период).

Также предлагаю доработать сервис загрузки истории для скачивания, чтобы можно было выбрать не только период скачивания, но и географию - на основании отображаемого на экране региона. Сам регион уже распознается сервисом (указание на текущий город в верхней строке сайта).
[image]

tubleronchik, 2022-10-20:
@vol4tim посмотри, пожалуйста.

positivecrash, 2023-06-12:
Сделали

closed 2023-06-12 by positivecrash