airalab/sensors.social #92: Объединить населенные пункты, если возможно

yante70, 2022-11-22:
У нас в Новороссийске и пригородах установлены датчики, но на разных картах они числятся в разных населенных пунктах - Гайдук, Глебовское, Южная Озерейка, Большие Хутора, Мысхако, и др.
  Но на самом деле это пригороды Новороссийска, и хотелось бы видеть из "в городе", а не перелистывать кучу "населенных пунктов"...
  Вот еще карта https://data.airtube.info/table.php?source=&country=RU&city= , к которой можно добавить название "наших"  населенных пунктов, и она покажет информацию по датчикам. установленных в каждом из них...
Novorossiysk
Myskhako
Yuzhnaya+Ozereevka
Glebovskoe
Gaiduk
Bol'shie+Khutora

  Да. у вас ни сайте список этих пунктов можно найти по ссылке https://sensors.robonomics.network/#/download

  С уважением, Ян, yante@yandex.ru Новороссийск

vol4tim, 2022-11-23:
Подумаем, как это можно будет сделать.

yante70, 2022-12-11:
Получается что-то?

yante70, 2023-06-12:
Хотелось бы узнать, насколько продвинулось решение моего вопроса...
  Мысхако, Борисовка и Ю.Озерейка - это все небольшие ( до 10тыс ) населенные пункты вокруг Новороссийска. И хотелось бы, чтобы они НЕ были отдельно...
  Или чтобы была возможность качать их либо отдельно, либо в составе нашего Города...

yante70, 2023-06-14:
А тут пока тишина...
  Напоминаю, что Борисовка, Владимировка, Глебовское, Мысхако, Южная Озерейка - это все пригороды Новороссийска, и их не обязательно выводить как отдельные населенные пункты :)

vol4tim, 2023-06-16:
Все датчики, из этих населенных пунктов, объединили с Новороссийском.

closed 2023-06-16 by vol4tim

yante70, 2023-06-16:
Спасибо, позже гляну...