airalab/sensors.social #94: Смена интерфейса

yante70, 2022-12-01:
Я вижу, сменили интерфейс, и даже попробовал "ночную схему"
  Так вот попробуйте и вы, выбрав "ночную схему" нажать на кнопку "скачать историю" и попробовать выбрать Город.
  Ну что, видно названия городов? :)

positivecrash, 2023-06-12:
[image]

выбор названий городов при скачивании истории - это ваша настройка устройства. Пришлите, пожалуйста, скриншот, если городов не видно

tubleronchik, 2023-06-12:
@positivecrash у меня такая же проблема. 

[image]

positivecrash, 2023-06-13:
@tubleronchik какая ОС и браузер?

tubleronchik, 2023-06-13:
@positivecrash Ubuntu 20.04, Chrome & Brave. В Firefox все ок

yante70, 2023-06-13:
Я бы не сказал, что в FireFox все ОК, но получше, чем в Хроме,,,

positivecrash, 2023-06-14:
> Я бы не сказал, что в FireFox все ОК, но получше, чем в Хроме,,,
> Хотя нет, то же самое - я в первый раз смотрел на своем компе по удаленке, а сейчас глянул уже будучи дома - нет разницы...
> Скрины нужны?

Скрины пока не нужны, спасибо! Проверьте, пожалуйста, теперь города должны быть видны

yante70, 2023-06-14:
> 

  Вопрос видимости городов при ночной схеме - снимается...
  Но остается вопрос объединения наших пригородов - см. другую тему... :)

positivecrash, 2023-06-15:
> Вопрос видимости городов при ночной схеме - снимается...
> Но остается вопрос объединения наших пригородов - см. другую тему... :)

супер, спасибо за ваше участие! По объединению городов мой коллега смотрит, я тут не могу ничего сказать :)

closed 2023-06-15 by positivecrash

yante70, 2023-06-15:
> > Вопрос видимости городов при ночной схеме - снимается...
> > Но остается вопрос объединения наших пригородов - см. другую тему... :)
> 
> супер, спасибо за ваше участие! По объединению городов мой коллега смотрит, я тут не могу ничего сказать :)

  Что-то ваш коллеги долго смотрит...