airalab/sensors.social #98: Как нужно настроить чужие датчики, чтобы они могли слать данныее на sensors.robonomics.network

yante70, 2022-12-09:
У меня есть датчик не вашей сборки, который шлет данные на sensor.community
  Как нужно в нем прописать ваши настройки, чтобы он мог слать данные и на ваш сайт?
  Спасибо

tubleronchik, 2022-12-29:
Привет. Подскажите, пожалуйста, какой датчик используется.

yante70, 2022-12-30:
Ну, датчики-то стандартные - РМ25/РМ10 и температура/влажность, я имел ввиду сборку...
  Сборки у нас две разные - Тольятти ( шлет данные на свой сайт и на Сенсор.Комунити ) и Краснодар ( только на последний )
  И вот вопрос - что нужно добавить в Краснодар, чтобы он добавился на карте Тольятти?

tubleronchik, 2023-02-07:
Привет!
В теории можно добавить наш адрес (например сервер [2.connectivity.robonomics.network](http://2.connectivity.robonomics.network/), порт 65). На картинке показан пример добавления нашего сервера в оригинальную прошивку.
[image]

Но тогда должен быть подключен датчик gps, тк оригинальная прошивка, в отличии от нашей, не позволяет вписать координаты GPS через интерфейс. Кроме того, наша прошивка меняет сервера, то есть в случае проблем с одним, данные будут передаваться на дргугие доступные сервера.
Если перепрошивать, то вот инструкция. С этой прошивкой данные также будут отправляться на [sensor.community](http://sensor.community/) https://wiki.robonomics.network/docs/sensor-setup/#requirements.

closed 2023-06-12 by tubleronchik