airalab/sensors.social #112: "Скачать историю"

yante70, 2023-06-19:
Сейчас у вас выгружается все в CSV-формате, а можно попросить в качестве альтернативы выгружать и в JSON?
  Структуру, например, можно взять на Сенсор.Комьюнити ...

yante70, 2023-07-01:
Спасибо, что за две недели совсем не прореагировали :)

positivecrash, 2025-07-01:
This issue is deprecated, but not forgotten. Related (in progress):
https://github.com/airalab/sensors.social/issues/221
https://github.com/airalab/sensors.social/issues/219

RU: Это ишью закрываю так, как ого устарело, но не забыто (разбито на две задачи, они в процессе решения)

yante70, 2025-07-01:
Закрывайте, конечно...
  А когда сделают, можно узнать?
  Хотя, сейчас я уже умею читать у вас в JSON, вопрос только во времени - мы живем по МСК, а там качается по Гринвичу, и получается немного фигня...
  Хотя, я в адресной строке формирую именно время, и можно было бы как-то качать по МСК...
  Подумаете? :)